openLuat/LuatOS #64: ubuntu 环境编译bsp/linux下demo后执行./luatos 提示找不到sys

yougaliu007, 2022-01-20:
如题，在ubuntu环境下载luatOS代码后，编译bsp/linux下代码，执行如下命令，提示找不到sys模块。
luatos可执行文件所在目录 bsp/linux/build

$ ./luatos -i ../main.lua
Lua 5.3.6  Copyright (C) 1994-2018 Lua.org, PUC-Rio
Enhance by LuatOS
E/lua.err ./luatos: 
E/lua.err ../main.lua:3: module 'sys' not found
stack traceback:
        [C]: in global 'require'
        ../main.lua:3: in main chunk
        [C]: in ?

chenxuuu, 2022-01-20:
确实，你可以手动把`sys.lua`复制到当前目录

yougaliu007, 2022-01-20:
copy sys.lua and main.lua file to luatos dir, still can not find sys

~/iot/project/luart/LuatOS/bsp/linux/build$ tree -L 1
.
├── CMakeCache.txt
├── CMakeFiles
├── cmake_install.cmake
├── hello.lua
├── libcjson.a
├── libfreertos10.a
├── liblua.a
├── libluat.a
├── libluatos_msys.a
├── luatos
├── main.lua
├── Makefile
├── mbedtls.out
└── sys.lua

2 directories, 12 files
~/iot/project/luart/LuatOS/bsp/linux/build$ ./luatos -i main.lua 
Lua 5.3.6  Copyright (C) 1994-2018 Lua.org, PUC-Rio
Enhance by LuatOS
E/lua.err ./luatos: 
E/lua.err main.lua:3: module 'sys' not found
stack traceback:
        [C]: in global 'require'
        main.lua:3: in main chunk
        [C]: in ?

chenxuuu, 2022-01-20:
https://github.com/openLuat/LuatOS/blob/master/lua/src/loadlib.c#L502-L533

我觉得是这里导致的问题

chenxuuu, 2022-01-20:
@wendal

yougaliu007, 2022-01-20:
~/iot/project/luart/LuatOS/bsp/linux/build$ ls / | grep lua
sys.lua

~/iot/project/luart/LuatOS/bsp/linux/build$ ./luatos -i main.lua 
Lua 5.3.6  Copyright (C) 1994-2018 Lua.org, PUC-Rio
Enhance by LuatOS
I/user.sys      from linux
E/user.coroutine.resume main.lua:11: attempt to index a nil value (global 'lvgl')
stack traceback:
        main.lua:11: in function <main.lua:7>
E/lua.err ./luatos: 
E/lua.err /sys.lua:370: main.lua:11: attempt to index a nil value (global 'lvgl')
stack traceback:
        main.lua:11: in function <main.lua:7>
stack traceback:
        [C]: in local 'cb'
        /sys.lua:370: in upvalue 'safeRun'
        /sys.lua:393: in function 'sys.run'
        main.lua:19: in main chunk
        [C]: in ?

wendal, 2022-03-06:
需要编译成32位才能兼容

wendal, 2022-04-14:
因为内嵌的sys.luac是32位的

closed 2022-04-14 by wendal