openLuat/LuatOS #83: Source for soc_download

racerxdl, 2023-01-01:
Hi! I'm making some stuff for AIR105 MCU and I would like to know where I can get the source code for the soc_download.exe programmer.

My idea is to make a Linux port of it, so its easier for linux developers to use it.

Is it available somewhere?

Thanks!

wendal, 2023-01-02:
ok, i will check

wendal, 2023-01-03:
[soc_download.zip](https://github.com/openLuat/LuatOS/files/10336191/soc_download.zip)

check this, need vs 2015 with win7 sdk

closed 2023-01-03 by wendal

racerxdl, 2023-01-16:
Not sure if it's in your interest, but I did a small tool for uploading in any platform: https://github.com/racerxdl/air105-uploader

It also supports code-signing for secure boot, which I did reverse from the original megahunt ISP Tool. I'm still need to add the secure-boot enabler though.

wendal, 2023-01-20:
it is great !