openLuat/LuatOS #89: [Question] How to convert a lua script to script.bin without windows and luatools?

cjacker, 2023-02-26:
As topic.

luatools is close source and only for windows, it can not run with wine on Linux.

I can generate the bootloader.bin and app.bin using xmake for specific SOC, for example, air105.

But how to convert a lua script to script.bin without luatools?

cjacker, 2023-02-26:
看了一下luadb v2的定义：

https://wiki.luatos.com/develop/contribute/luadb.html

尝试理解 `luat/vfs/luat_luadb_inline_sys.c`, 跟SPEC还是有些不同：
```
const char luadb_inline_sys[] = {

0x01, 0x04, 0x5A, 0xA5, 0x5A, 0xA5,  // 跟SPEC不同，SPEC讲"值为 0xA5 0x5A 0xA5 0x5A", 这里是逆序？

0x02, 0x02, 0x00, 0x02, // OK

0x03, 0x04, 0x12, 0x00, 0x00, 0x00, // 逆序？如果按SPEC，这个值必然是0x12（十进制18），不可能存在其它值。0x01行 len = 6，0x02行 len = 4，0x04行 len = 4, 0xFE行 len = 4.

0x04, 0x02, 0x01, 0x00, //逆序？1个文件。
0xFE, 0x02, 0xFF, 0xFF, // 哪些内容的CRC16值？

0x01, 0x04, 0x5A, 0xA5, 0x5A, 0xA5, 逆序？
0x02, 0x08, 0x73, 0x79, 0x73, 0x2E, 0x6C, 0x75, 0x61, 0x63, //正序？sys.luac。
0x03, 0x04, 0xB9, 0x15, 0x00, 0x00, // 长度，逆序？按spec，接下来是文件内容。但接下来是0xFE.
0xFE, 0x02, 0xFF, 0xFF, //同样的问题，谁的CRC16校验值?

// 后面应该是文件内容开始
0x1B, 0x4C, 0x75, 0x61, 0x53, 0x00,
0x19, 0x93, 0x0D, 0x0A, 0x1A, 0x0A, 0x04, 0x04,
0x04, 0x04, 0x04, 0x78, 0x56, 0x00, 0x00, 0x00,
0x40, 0xB9, 0x43, 0x01, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x15,
...
```

最后，按SPEC, "文件区的末尾, 会是一个名为 “.airm2m_all_crc#.bin”, 内容是当前文件数据之前的所有数据的md5的hex值."，文件名应该是正序，"0x2E,0x61,0x69,0x72,0x6D,0x32,0x6D,0x5F,0x61,0x6C,0x6C,0x5F,0x63,0x72,0x63,0x23,0x2E,0x62,0x69,0x6E"，但luadb_inline_sys中并没有这个值序列。

能够提供完整准确的luadb v2 spec定义？

wendal, 2023-02-27:
1. `.airm2m_all_crc#.bin` 是Air724是为了校验luadb多加的文件, 对luatos来说没有意义
2. 内置的luadb数组已经废弃, 而且不需要`.airm2m_all_crc#.bin`文件, 因为用不到, 其实不校验的
3. wiki里的描述就是准确的

基本流程就是
1. lua 转 luac, 因为luatos区分32bit/64bit两个版本, 所以需要用不同的luac程序
2. luac转luadb, 就是一个头部 + N个文件的文件体, 实际不限于luac文件, 也不要求`.airm2m_all_crc#.bin`文件
3. 在最近的适配中, 已经支持romfs格式, 后续可能会推广使用, 但依然会兼容luadb格式

wendal, 2023-02-27:
https://github.com/openLuat/LuatOS/blob/master/bsp/win32/tools/update_inline_sys.lua

这个文件里还有些残余, 注释的部分, 稍作修改应该就能跨平台使用

cjacker, 2023-02-27:
从Linux的角度看，xmake和luatOS的构建都没有任何问题。

目前仅有几个工具是欠缺的:

- script.bin的制作工具，或者您所提到的后续romfs的制作工具。有了这个才能将脚本做成一个binary烧到特定的位置。
- air101/103的flash工具。air101_flash.exe不开源，xt804有wm_tool，开源，在linux能用，问题不大，总可以烧进去。
- air105的烧写工具。air105换成了mh1903s ARM，本身应该可以用CMSIS-DAP。可能出于推广的考虑，官方的AIR105参考板把SWD引脚藏起来没引出。这就比较尴尬了，在不动板子硬件的情况下，没有一个能用的工具可以在Linux烧写。

以下仅是个人建议：

1，bootloader/app烧写完成后，script部分是架构基本无关的，可以有一个单独通用luatos script flash tool。不升级bl/app的情况下，开发时更多是烧脚本，没必要弄的太复杂，GUI工具很难自动化，远不如一条命令来的方便，windows也是一样。

2，架构相关部分的烧写，继续现在的luatools之外(是否考虑Linux version)，是否考虑提供最基本的Linux cmdline tool或倚靠架构本身的工具完成bl和app的烧写，如果是ARM，那工具实在太多了，没必要屏蔽或刻意避免使用这些现成的工具。比如air105在linux下的问题，除了没有一个脚本转bin的工具外，就是硬件swd引脚没引出，否则总可以用cmsis dap烧进去。

cjacker, 2023-02-27:
顺便，我调了一下soc_download的代码，甚至修改了部分，用win10 sdk和msvc toolchain在linux下重新编译了一遍(可以过，没问题)，但PortRW始终超时，严重怀疑是不是[rts linux和windows实现不一样](https://stackoverflow.com/questions/5090451/how-to-open-serial-port-in-linux-without-changing-any-pin)的问题。这里也提到了"[Linux下的问题和解决](https://www.cnblogs.com/milton/p/15621540.html)"。但即使改了内核ch341驱动，也不行。air105-uploader这个[issue](https://github.com/racerxdl/air105-uploader/issues/1)也很有意思。

自己也写了个python脚本试了下，不行。

简单点说，就是reset后，不论发20个或更多个0xf8过去，始终进不了bootloader。

真不知道问题在哪了？要不然，写个uart烧写脚本也并不难。

wendal, 2023-02-28:
但提供的soc_download在windows下是ok的吧?

wendal, 2023-02-28:
1. script.bin的生成, 晚点提供一个脚本吧
2. air101_flash 就是wm_tools小改了一下复位逻辑, 其他没大区别的, 需要的话待会上传一下源码
3. air105的swd的确是失误了,只能等这批开发板快卖完了, 新版改pcb了
4. 有计划 https://github.com/LuatOS-Downloader , 但当前的重点EC618没有上游提供的下载协议, 只能逆向, 就很蛋疼

wendal, 2023-02-28:
2. air101_flash的源码是有的 https://github.com/openLuat/luatos-soc-air101/blob/master/tools/xt804/wm_tool.c

cjacker, 2023-02-28:
> 2. air101_flash的源码是有的 https://github.com/openLuat/luatos-soc-air101/blob/master/tools/xt804/wm_tool.c

OK，我之前一直用wm_sdk试的。那对XT804来说，要把luatos开发用起来，现在只缺一个script.bin或romfs的工具了。

对air105来说，air105-uploader问题已经修复了，可以使用mh1903或者air105 firmware library来写东西了。暂时还不能program luatos。

cjacker, 2023-02-28:
> 1. script.bin的生成, 晚点提供一个脚本吧
> 
>     2. air101_flash 就是wm_tools小改了一下复位逻辑, 其他没大区别的, 需要的话待会上传一下源码
> 
>     3. air105的swd的确是失误了,只能等这批开发板快卖完了, 新版改pcb了
> 
>     4. 有计划 https://github.com/LuatOS-Downloader , 但当前的重点EC618没有上游提供的下载协议, 只能逆向, 就很蛋疼

air101_flash代码传上来吧。我在写w80x/air101/103的教程，搞定luatos在linux下的开发后会把这块内容补上。

https://github.com/cjacker/opensource-toolchain-w80x

wendal, 2023-02-28:
https://github.com/openLuat/LuatOS/blob/master/tools/luadb/luadb_maker.lua

wendal, 2023-03-10:
ok了吗? 没问题就close了

cjacker, 2023-03-10:
先不用关。

luadb_maker可能是有问题的。比如:
TLD(buff, 0x02, value)
TLD(buff, 0x03, io.fileSize(script_dir .. "\\" .. value))
这个value的长度一定是2或者4吗？

另，今天有时间，用C写了个luatos-utils，您在windows下也试一下，主要是mkscriptbin.c能不能在win下编译和正常使用。

https://github.com/cjacker/luatos-utils

接下来我考虑把esp32和air105的支持加上去。

cjacker, 2023-03-10:
另外，luac用gcc -m64编译，即使打开了-DLUA_32BITS也是有问题的，跟gcc -m32编译的luac生成的main.luac只有一点区别：

gcc -m32编译的luac生成的头 : `1b 4c 75 61 53 00 19 93  0d 0a 1a 0a 04 04 04 04`
gcc -m64 编译的luac生成的头 : `1b 4c 75 61 53 00 19 93  0d 0a 1a 0a 04 08 04 04`

注意倒数第3位。剩余内容均没有区别。

但08在luatos是不支持的，是否可以考虑加上去？

allewalker, 2023-03-21:
> 顺便，我调了一下soc_download的代码，甚至修改了部分，用win10 sdk和msvc toolchain在linux下重新编译了一遍(可以过，没问题)，但PortRW始终超时，严重怀疑是不是[rts linux和windows实现不一样](https://stackoverflow.com/questions/5090451/how-to-open-serial-port-in-linux-without-changing-any-pin)的问题。这里也提到了"[Linux下的问题和解决](https://www.cnblogs.com/milton/p/15621540.html)"。但即使改了内核ch341驱动，也不行。air105-uploader这个[issue](https://github.com/racerxdl/air105-uploader/issues/1)也很有意思。
> 
> Update: air105-uploader的问题已经修复了.

PortRW用了win api的异步通讯机制，转到linux下的话，可以用select，另外RTS最多拉高拉低不一样，大不了就手动重启板子，一样可以下载的

jinguo-yin, 2023-12-31:
实在不行就直接在CSDK中自己开发一个支持LUA虚拟机的任务，抛开官方的LUA支持。这个工作量反而更小

wendal, 2024-05-16:
https://github.com/wendal/flashtoolcli4ec
https://github.com/openLuat/luatos-soc-pc

均可了

closed 2024-05-16 by wendal