openLuat/LuatOS #99: [🐛Bug]: Air601使用AirM2M_601_V1001_WIFI_AT.soc固件，AT指令部分功能有问题

YY1238, 2023-11-17:
### 描述一下这个bug / Describe the bug

1、使用AT指令修改默认波特率，复位后仍然不生效
2、配置air601为TCP服务器，发送100多次后再发送就失败了，但是发送失败情况发生前后，接收一直正常

### 复现步骤 / To Reproduce

1、使用AT指令修改默认波特率，对应指令为 AT+UART_DEF=9600,8,1,0,0，正常响应OK，但是复位后使用9600波特率无法通讯，仍然得使用115200波特率才有反馈，并且查看默认波特率配置是有修改为9600
2、配置air601为TCP服务器，正常连接上后间隔200ms间隔自动发送数据给对应客户端，前面114次正常发送，客户端也是正常接收，但是114次后air601发送显示失败，但是air601出现发送失败的情况后，手机客户端发送的数据仍然可以正常接收，打印在串口助手上，并且是客户端实时的数据，复位601，重新配置tcp服务器后依旧是前面的114次正常，往后就发送失败

### 如果正常，应该是什么样 / Expected behavior

[image]

### 截图 / Screenshots

[image]

### 日志 / Logs

无

### PACK包版本 / Version

AirM2M_601_V1001_WIFI_AT.soc

### 验证

- [X] 检查过该问题，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.
- [X] 已经提供了完整的报错信息、日志、截图，没有经过删减。

wendal, 2023-11-24:
似乎是AT固件与LuatOS固件在系统参数的冲突

YY1238, 2023-12-01:
@

closed 2024-09-11 by wendal