openLuat/LuatOS #101: nimble.disconnect()无法断开蓝牙连接

mxdljwxx, 2023-11-23:
### 描述一下这个bug / Describe the bug

LuatOS-SoC_V1022_AIR601,  nimble.disconnect()不能断开蓝牙连接！
查luatos源码，luat_nimble_mode_central.c 中
`if (!g_ble_conn_handle) {   
   return 0;
}`

应全局连接g_ble_conn_handle = 0 ,导致if判断语句为1，退出，不执行断链语句。
可能在lua中编程习惯，C中只有0和1。
建议修改为
`if (g_ble_conn_handle < 0)`

### 复现步骤 / To Reproduce

如上

### 如果正常，应该是什么样 / Expected behavior

主动断开连接

### 截图 / Screenshots

[image]

### 日志 / Logs

如图

### PACK包版本 / Version

LuatOS-SoC_V1022_AIR601

### 验证

- [X] 检查过该问题，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.
- [X] 已经提供了完整的报错信息、日志、截图，没有经过删减。

wendal, 2023-11-24:
-_- 下周看看

wendal, 2023-11-24:
咦, 已经分析出原因了? 直接给个pull request嘛

closed 2023-12-08 by wendal