openLuat/LuatOS #102: [🚀新需求]: Luatool在macos的windows虚拟机环境下无法烧录，ectools在ubuntu虚拟机中正常工作，有没有办法使用ectool烧录lua代码？

jinguo-yin, 2023-12-31:
### 您的功能请求是否与解决某些问题有关？请描述一下。/ Is your feature request related to a problem? Please describe.

Luatool在macos的windows虚拟机环境下无法烧录，ectools在ubuntu虚拟机中正常工作，有没有办法使用ectool烧录lua代码？

### 描述您想要的解决方案 / Describe the solution you'd like

在虚拟机中使用luatool

### 描述您想要的详细使用步骤描述 / Describe the solution you'd like to use in what way

在虚拟机中使用luatool

### 确认信息

- [X] 检查过该需求，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 这个需求是真实存在的，并且对于本项目来说是必要的
- [X] 我已经提供了该需求的清晰、详细、完整的方案，请按该方案进行实施
- [X] 我认为我提供的信息已经足够，不需要其他人继续完善该需求的其他细节
- [X] 我同意跟进该需求的实施进程，并在适时提供合理的解释与纠正

chenxuuu, 2024-02-04:
你需要自行打包好固件文件，再用ectools烧录

easylolicon, 2024-06-11:
我在macos中使用 pd（ParallelsDesktop）虚拟机，语法验证都没问题，烧录时提示 下载失败：下载失败

spinachlover, 2024-08-05:
> 我在macos中使用 pd（ParallelsDesktop）虚拟机，语法验证都没问题，烧录时提示 下载失败：下载失败

相同问题，linux下的vbox win虚拟机，显示下载失败，ectool在linux下正常