openLuat/LuatOS #105: [🚀新需求]: FIBOCOM L610

tcpipchip, 2024-09-16:
### 您的功能请求是否与解决某些问题有关？请描述一下。/ Is your feature request related to a problem? Please describe.

Not exactly a problem

### 描述您想要的解决方案 / Describe the solution you'd like

I would like to know if is possible to run the LUATOS on RDA8910 FIBOCOM L610-GL

### 描述您想要的详细使用步骤描述 / Describe the solution you'd like to use in what way

I can run the examples to AIR72X...but no NETWORK connection

### 其他备注信息或截图 / Add any other context or screenshots about the feature request here

[I]-[ril.defrsp] AT+EEMGINFO? true OK nil
[I]-[ril.sendat] AT+CSQ
[I]-[ril.proatc] +CSQ: 99,99
[I]-[ril.proatc] OK
[I]-[ril.sendat] AT+CESQ
[I]-[ril.proatc] +CESQ: 99,99,255,255,255,255
[I]-[ril.proatc] OK
[I]-[ril.sendat] AT+EEMGINFO?
[I]-[ril.proatc] +EEMGINFO: 3, 2
[I]-[ril.defurc] +EEMGINFO: 3, 2
[I]-[ril.proatc] OK
[I]-[ril.defrsp] AT+EEMGINFO? true OK nil
[I]-[ril.sendat] AT+CSQ
[I]-[ril.proatc] +CSQ: 99,99
[I]-[ril.proatc] OK
[I]-[ril.sendat] AT+CESQ
[I]-[ril.proatc] +CESQ: 99,99,255,255,255,255
[I]-[ril.proatc] OK
[I]-[ril.sendat] AT+EEMGINFO?
[I]-[ril.proatc] +EEMGINFO: 3, 2
[I]-[ril.defurc] +EEMGINFO: 3, 2
[I]-[ril.proatc] OK
[I]-[ril.defrsp] AT+EEMGINFO? true OK nil

### 确认信息

- [X] 检查过该需求，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 这个需求是真实存在的，并且对于本项目来说是必要的
- [X] 我已经提供了该需求的清晰、详细、完整的方案，请按该方案进行实施
- [X] 我认为我提供的信息已经足够，不需要其他人继续完善该需求的其他细节
- [X] 我同意跟进该需求的实施进程，并在适时提供合理的解释与纠正

wendal, 2024-10-08:
做不到

closed 2024-10-08 by wendal

tcpipchip, 2025-01-29:
some guys said that not works on SOUTH AMERICA the LuatOS