openLuat/LuatOS #110: [Question] How to flash Air724UG on linux? LuatOS or LuatOS-Air?

subado, 2025-04-06:
Hello, I don't understand how to flash firmware without Luatools on linux. Can I somehow flash firmware without Luatools? Can I use or modify this repo https://github.com/cjacker/luatos-utils to achieve my goal? Is LuatOS-Air deprecated? Does LuatOS support Air724UG?

wendal, 2025-04-08:
LuatOS-Air isn't deprecated, but not active develop, we will fix bug .

The original mass production tools can also flash the machine, but the synthesized pac file needs to be flashed.

wendal, 2025-04-08:
LuatOS for Air724UG, it is suspend development, but almost done, download from 

https://www.air32.cn/air724ug/

subado, 2025-04-09:
Can I have a link to 
> the the original mass production tools
?

closed 2025-11-13 by wendal