openLuat/LuatOS #113: [🚀新需求]: 请求disp库设置时钟极性(CPOL)

skirky3605, 2025-07-29:
### 您的功能请求是否与解决某些问题有关？请描述一下。/ Is your feature request related to a problem? Please describe.

我们的项目需要使用Air724ug模块驱动一块st7565r芯片的单色屏幕，st7565r要求cpol为0，但是disp库默认cpol似乎是1，文档中并没有给出关于disp如何调整cpol

### 描述您想要的解决方案 / Describe the solution you'd like

希望可以添加一个初始化参数使得disp初始spi的cpol可以自行调整（最好带上cpha）

### 描述您想要的详细使用步骤描述 / Describe the solution you'd like to use in what way

无

### 其他备注信息或截图 / Add any other context or screenshots about the feature request here

无

### 确认信息

- [x] 检查过该需求，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [x] 这个需求是真实存在的，并且对于本项目来说是必要的
- [x] 我已经提供了该需求的清晰、详细、完整的方案，请按该方案进行实施
- [x] 我认为我提供的信息已经足够，不需要其他人继续完善该需求的其他细节
- [x] 我同意跟进该需求的实施进程，并在适时提供合理的解释与纠正

wendal, 2025-08-05:
luatos-air系列固件不会新增功能了

skirky3605, 2025-08-06:
那修改disp的时钟极性无法通过其它软件手段实现了吧